VoronDesign/Voron-Stealthburner #3: BOM missing

ScottSmall, 2022-07-24:
Quick skim, not seeing BOM in SB repo or as tab in Sourcing Guide

codegrunt, 2022-07-31:
https://vorondesign.com/voron_stealthburner -> click on bom

BerzerkaDurk, 2022-08-03:
> https://vorondesign.com/voron_stealthburner -> click on bom

this all points at the Beta. The SB repo needs one, too.

codegrunt, 2022-08-03:
The stls, yes. the bom however is located on the site for all voron
projects last I checked.  If you don't trust that bom I can't help you
further.

BerzerkaDurk, 2022-08-05:
It's not a matter of trust. The link you gave points to an outdated BoM that is no longer accurate. I appreciate that you were trying to help the person who created the issue. My point was simply that the BoM still needs to be added to the repo.

wile-e1, 2022-08-18:
> It's not a matter of trust. The link you gave points to an outdated BoM that is no longer accurate. I appreciate that you were trying to help the person who created the issue. My point was simply that the BoM still needs to be added to the repo.

I have submitted a PR for the VoronDesign site to update the BOM.

As for the BOM being added to the repo, we don't add BOMs to our repos because we have them on the main site.  Having the same information in too many places means that there's more likelihood that things won't match and it's twice as much work to keep things updated.

closed 2022-08-18 by wile-e1