VoronDesign/Voron-Stealthburner #21: Tool Cartridge and Dragon Hotend

narf222, 2022-07-31:
hi all,

just a small issue: on page 42 (manual from 2022-07-30) you mention M3x8 bolts to hold the hotend in place. the dragon hotend needs M2,5 x8mm. others won't fit the hotend but also don't find your printed part by design. the screw head would be to big to go into the hole of the printed part.

so, i'd suggest to write that at least in the annotations for the printheads (here: https://github.com/VoronDesign/Voron-Stealthburner/tree/main/STLs/Stealthburner/Printheads ) or in the manual, too. the thing is that some people may need to buy those screws extra (because M2,5 aren't part of any BOM). that's why it's an issue worth mentioning. 

thanks for everything you awesome people :))

cheers
narf

bassco, 2022-08-01:
The M2.5x8mm bolts are supplied with the Dragon Hotend and are therefore not required in the BOM. 

Agreed, that an annotation would be useful to clear this up.

wile-e1, 2022-08-02:
The M3x8 are for the Revo Voron toolhead shown in the manual,  hence the note "M3 Mounting Bolts?  Cause we can, that's why"

I don't think it's realistics to detail every different toolhead assembly.  Perhaps we could make it more clear that M3x8 are for RevoVoron ONLY.

narf222, 2022-08-02:
thank you for your answers. and yes, there were 2 bolts M2.5 included with the dragon hotend, but not 4. And even those 2 were probably 6 instead of 8mm (haven't measured them, though)

PR #27 (2022-08-16): Manual update
page 42 - added note that other hotends might need different hardware
page 67 - added callout for bolt length

closes #21
closed 2022-08-16 by raymondh2