VoronDesign/Voron-Stealthburner #24: Untracked design changes

GarthSnyder, 2022-08-04:
As of 5a2a1fd, there seem to have been minor modifications to the CW2 latch (relative to RC1) that are not mentioned in the logs and not reflected in STL changes. I didn't check other elements, just happened to be looking at the latch. 

Is there a system in place to collect this information, or does someone have to manually go through and manually diff the whole design at the next release?

nemgrea, 2022-08-04:
could you be more specific? this is the current stl and an stl created from the current cad and im not seeing the modifications you are referring to
[image]

GarthSnyder, 2022-08-04:
Sure, here are some additional details. Just to be clear, I am comparing CAD/Clockwork2.stp from e6cc7d7 (aka RC1) with the same file as of the now-current HEAD, 5a2a1fd.

Here's a GIF animation showing the differences. It looks like this was intended as a clean-up, but it does modify the thickness of the struts in several regions.

[image]

Here are the CAD bodies superimposed in Fusion 360:

[Superimposed latches.f3d.zip](https://github.com/VoronDesign/Voron-Stealthburner/files/9262553/Superimposed.latches.f3d.zip)

nemgrea, 2022-08-04:
i guess I'm confused by what you are asking? is there a change that GitHub has not captured? or are you expressing the desire for a written list of changes beyond the commit history of the components?

GarthSnyder, 2022-08-04:
As a client of this repo that will be maintaining downstream mods, I would like to be able to find out whether parts of interest to me have changed. More broadly, everyone who prints RC1 is going to want to know what to reprint for the actual release. I worry that this information is getting lost in the shuffle.

Currently, the only way to determine changes is to manually import the STEP file into a CAD system and check each part by hand, every time there is a commit.

I guess that what I would really like to see is consistency between STLs and STEP. If the CAD is changed, the STLs should change. I don't actually care about the STLs themselves, but because they break everything out at the part level, they make it easy to determine what parts have changed between any two commits using git.

nemgrea, 2022-08-04:
ok we're back to talking about different things again here...you specifically showed a gif of stl to stl comparison and now your talking about a step to stl comparison? 

i agree with you if there is a change to the cad then you should likely see a change to the corresponding stl file, only in very rare cases would this not be true, i believe we are in agreement here, 

what im not following though is whether you observe an issue with the CURRENT stls vs the CURRENT steps? 

im not sure why you are expecting RC1 files to be the same as the current files when you can see in the commit history that both the cad and the stls have hade newer files uploaded? if a new file is uploaded it can be assumed that it has changed, the extent of that change should be mentioned in the commit notes.

GarthSnyder, 2022-08-04:
> ok we're back to talking about different things again here...you specifically showed a gif of stl to stl comparison and now your talking about a step to stl comparison?

This is just getting silly. I don't know how to make it any more explicit than "Just to be clear, I am comparing CAD/Clockwork2.stp from https://github.com/VoronDesign/Voron-Stealthburner/commit/e6cc7d7b1d68a9c1707e0b089f6d283f2d530d41 (aka RC1) with the same file as of the now-current HEAD, https://github.com/VoronDesign/Voron-Stealthburner/commit/5a2a1fd0f86acc9204232f64b62abddbb71ca22a." The image obviously does not show STL bodies.

> what im not following though is whether you observe an issue with the CURRENT stls vs the CURRENT steps?

**YES**. The current STLs/Clockwork2/[a]_latch.stl file does match the CAD. That file has not been modified since RC1.

> im not sure why you are expecting RC1 files to be the same as the current files when you can see in the commit history that both the cad and the stls have hade newer files uploaded?

Some STLs have changed along with their CAD models and, in at least one case, others have not. Does this not seem like cause for concern?

nemgrea, 2022-08-04:
There have been a number of commits since the rc1 files have been released, that rc1 release you are linking is Not the current stl files.

The release is only current at the time it is posted. Once new commits are merged the release page is out of date until a new RC is put together

GarthSnyder, 2022-08-04:
OK, I give up.

nemgrea, 2022-08-04:
> > 
> **YES**. The current STLs/Clockwork2/[a]_latch.stl file does not match the CAD. That file has not been modified since RC1.

WRONG!!

go here, open the latch.stl that is in the repository at this very moment
[image]
 look at the CURRENT latch stl file 
[image]

now go here and download the CURRENT cad file for the CW2
[image]
[image]
 and see that it matches the previously viewed stl file..
[image]

you are using an OLD latch file from the RC1 zip that has been updated and at this very moment BOTH the cad and the stl for that latch are identical...

raymondh2, 2022-08-04:
The release has not been updated. It is time consuming to do one with a change log every time we change a file.

closed 2022-08-04 by raymondh2