VoronDesign/Voron-Stealthburner #33: Of note...

Durahl, 2022-08-22:
Ever since the Beta release of the SB / CW2 Combo I was having an issue that would creep in as time would pass, sometimes after weeks, sometimes during the first few prints where prints would fail because of the already high friction in my ERCF equipped System ( 1.5m Bowden Length ) causing the Filament getting stripped - With the general increase in room temperature during the summer this issue seems to have skyrocketed pretty much breaking the final straw in getting nice or even just finished prints at all.

Most of the time I could fix this by reprinting the Guidler, Latch, and Shuttle parts hinting at the issue being _somewhere_ in that region.

- With the release of the SB / CW2 CAD Solids I found confirmation that the Active Drive Gear was situated in the shadow of the Filament Path - The Passive Guidler Gear would thus constantly push the Filament into the side of the Filament Path of that of the Active Drive Gear introducing unnecessary ( and in my case additional ) friction.
Shifting all of the Geometry surrounding the Active Drive Gear 0.3mm ( like those of the Bearing Pockets ) towards the Guilder is sufficient enough to fix that - The Filament will no longer be pressed against the Sides of the Filament Path.

- Another MOD I had to perform was fattening up the Guidler and Latch Parts as I'm suspecting one or the other to suffer from creep under the constant load while working under hightened temperatures:
[image]
Fattening up the Guidler and Lever reduces the range of motion of the latter but won't actually impact the release of the Shuttle - The changes are also barely noticeable from an aesthetiv viewpoint but using these mods appears to have remedied most of the issues I've been dealing with over the last few days.

- Of note is also that performing an Interference Check between the `StealthBurner_Body` and a `Toolhead` will show a conflict on either side of the now Angled Areas that prevent the Partcooling Air from Escaping via the gap before it makes it from the Fan to the Toolhead.

nemgrea, 2022-08-29:
regarding the last note. ` performing an Interference Check between the StealthBurner_Body and a Toolhead will show a conflict on either side of the now Angled Areas`

have you perhaps assembled your cad without the appropriate gap seen below?

[image]

Durahl, 2022-09-08:
That part with the under extrusion looks like that _because_ I was having those issues - I just barely managed to print _that_ one as an unbent stopgap before settling with the fattened one - No other parts I printed after using the fattened one looks like that.

**Concerning the Filament Path:**
Original unaltered - best-case scenario CAD Model with a 1.75mm Filament added into the Filament Path.
[image]
Filament touching the Active Drive Gear side of the Filament Path - Can't go any further than that with the Filament Path present at the crosssection.

[image]
Filament BARELY touching the Active Drive Gear.

[image]
Perspective view of Filament BARELY touching the Active Drive Gear.

The CAD shows you either have BARELY ( yea I'm totally running with it... ) any contact with the Active Drive Gear or the Filament gets pushed like mad into the Side Wall.

And again, that is a best-case scenario... The Real-World Scenario will have the Filament Path be even smaller in Diameter only increasing the issue.

wile-e1, 2022-09-08:
The drive gears in the CAD are modeled with their drive groove incredibly deep.  Much deeper than what is typical in reality for BMG drive gears.  Their minimum diameter is Ø7.0mm while in actuality, gears are generally Ø7.4mm at the smallest part.  So what you're actually showing here is not entirely accurate.  

CAD is always "best case scenario" and does not actually translate to reality.  The reality of the situation is that if this was indeed a systemic problem, nearly everyone should be seeing the same issues that you were seeing before your modifications.  

Myself and many many others on the dev team have been through dozens of iterations on these parts pre-beta, beta, and rc1, and no one has pointed fingers at the design for issues or suggested there is a catastrophic design fault that needs addressing.  I have printed everything from TPE to CF-ABS with CW2 over the past year and results have been the best I have ever seen since I started printing, so forgive my reluctance to jump in and start changing fundamental design parameters.

raymondh2, 2022-10-10:
Closing due to inactivity

closed 2022-10-10 by raymondh2