VoronDesign/Voron-Stealthburner #37: Wrong bolts length for tool cartridge in the manual

bistory, 2022-09-04:
Hi,

I followed the manual, everything looks good except for one step for the tool catridge (page **41**).
The manual tells to use M3x12 bolts but they won't fit.
It should be M3x14 (tried that, it works) or M3x16 (didn't try that but it should work, based on CAD).

Thanks for the hard work !

BerzerkaDurk, 2022-09-05:
I checked the toolhead models for all of the hotends, and the distance from the bottom of the counterbore to the opposite side of the rear mount is 10.5 mm on all of them, so a 12 mm screw should work fine.

Any chance your heatsets for those 2 screws are set below the surface of the front mount?

BerzerkaDurk, 2022-09-05:
[image]

bistory, 2022-09-05:
> [image]

Thanks for your check but it's not the same hole ^^

[image]

BerzerkaDurk, 2022-09-05:
That _is_ the right hole. It's the only hole with the hardware on the page you mentioned. Page 41

The way you are measuring shows the max depth the screw can go.

The way I'm measuring shows how much of the screw extends past the rear half. If the counterbore is 10.5 mm from the other side of the rear part, a 12 mm screw will extend 1.5 mm past and into the heatset in the front part. It's pretty basic.

bistory, 2022-09-05:
Ok strange then because my inserts are not 1.5mm below the surface...
I will check if my screws have the correct size

BerzerkaDurk, 2022-09-05:
Probably wouldn't hurt to change them to 14s anyway. Rule of thumb is usually 1 diameter of thread engagement minimum, and 1.5 mm only gives you 1/2 of a diameter.

wile-e1, 2022-09-07:
> Probably wouldn't hurt to change them to 14s anyway.
> 
> Rule of thumb is usually 1 diameter of thread engagement minimum, and 1.5 mm only gives you 1/2 of a diameter, plus you don't get a full thread all the way out to the end of the screw, further reducing engagement length.
> 
> Alternatively, the designers could, of course, keep the 12 mm screws and make the counterbores a bit deeper.

We won't be changing the bolt length to M3x14 because it is not a standard size in our Voron category of parts.

M3x12 work ok because the plastic under the head of the bolt compresses some when the screw is tightened.  
M3x16 also works because there is clearance on the other side of the insert to allow the screw to protrude past the heatset.

[image]

bistory, 2022-09-07:
Thanks for your reply @wile-e1 , I have checked again M3x12, it fits, now that the counterbore holes have been "done" by longer screws. The screws measure 12mm with no extra, the printer is correctly calibrated. I had no problem with other screws.

I think M3x16 wouldn't hurt because those pieces are not meant to be disassembled regularly...

Here is some pictures of how it fits with M3x12 : 

[image]
[image]

closed 2022-10-10 by raymondh2