VoronDesign/Voron-Stealthburner #40: BOM/Sourcing Guide observations

subatomicglue, 2022-09-22:
- [The Sourcing Guide](https://vorondesign.com/sourcing_guide?model=VSB) has a lot of empty cells, will it be updated at some point?
- In the [BOM](https://vorondesign.com/voron_stealthburner) the “Complete BMG Extruder Clone” link is dead at aliexpress (other link works) `Sorry, the page you requested can not be found:(`
- [The Sourcing Guide](https://vorondesign.com/sourcing_guide?model=VSB)'s “ADXL Optional” is not listed on the BOM (should it be?)
- I assume “ADXL Optional” is the ADXL 345 from google searches on it...  something like this?   https://www.amazon.com/Adafruit-ADXL345-Triple-Axis-Accelerometer-ADA1231/dp/B01BT4N9BC     or this (~6x cheaper)?   https://www.aliexpress.com/item/3256803808631787.html   many ADXL boards seem to have different form factors slightly...  how to choose, or identify what’s compatible?
-  [The Sourcing Guide](https://vorondesign.com/sourcing_guide?model=VSB)'s  Neopixel was missing links for sourcing, here's some I found:
  - https://www.aliexpress.com/item/3256804089586600.html
  - https://www.aliexpress.com/item/3256804145533673.html
  - hahah rainbow barf neopixels: https://kb-3d.com/store/printer-specific-harnesses/566-6854-rainbow-barf-led-toolhead-harness-for-stealth-burner-1653783045189.html#/279-connectortype-molex_microfit_30
- BOM/Sourcing:   Hotends you recommend for the proper stealthburner experience would be nice to see listed.
  - Is there a place where their merits and detriments are compared/discussed in a doc in re: to stealthburner?  The [README.md](https://github.com/VoronDesign/Voron-Stealthburner/tree/main/STLs/Stealthburner/Printheads) only has notes, but no guide to selecting them…   
    - Might be nice to link to this from the BOM/Sourcing guide to help people make their decisions...

Hope this helps RC1 get to Release :)   And I hope appropriate as a github issue :)

RealDeuce, 2022-10-11:
Just to add my two cents, the Amazon initeq inserts are really too big for the two heatsets by the big gear, I tried using them and had to cut the bulges off by the gear afterward.  The bottom one without the cutout was the worst, the heatset body actually extends past the edge even when it's cut back.

aeiounix, 2022-10-13:
> initeq inserts are really too big 
Or the holes are too small. OR holes are too big for the smaller inserts. I saw a related issue on the afterburner repo. The inserts that came with the insert tool I bought are too small for the holes. I figured this is one of those "we're all just people" type issues and I'm gonna push ahead and make it work or try again if it doesn't.

subatomicglue, 2022-10-13:
I haven't used my amazon initeq M3-0.5 heatsets yet...   but here's a picture:
[image]

subatomicglue, 2022-10-13:
For heatset inserts,  The stealthburner PDF manual links to the [The Voron sourcing guide](https://vorondesign.com/sourcing_guide), which lists :
```
M3 Brass Heatset Inserts - Short M3x5x4
```

And gives the following links for `M3xD5xL4`:
- [Aliexpress](https://www.aliexpress.us/item/2255800046543591.html?aff_fcid=81600eaeb0364e8da4684022dff7e6e2-1665677693572-03098-_ANjqIj&aff_fsk=_ANjqIj&aff_platform=portals-tool&sk=_ANjqIj&aff_trace_key=81600eaeb0364e8da4684022dff7e6e2-1665677693572-03098-_ANjqIj&terminal_id=7b5d55b6f519456dbcf1db2607edcaa6&afSmartRedirect=y&gatewayAdapt=glo2usa4itemAdapt&_randl_shipto=US)
- [Aliexpress](https://www.aliexpress.us/item/2255800046543591.html?aff_fcid=0c0bd9fe711348bb96b40f20d2dde394-1665677713570-09102-_ANjqIj&aff_fsk=_ANjqIj&aff_platform=portals-tool&sk=_ANjqIj&aff_trace_key=0c0bd9fe711348bb96b40f20d2dde394-1665677713570-09102-_ANjqIj&terminal_id=7b5d55b6f519456dbcf1db2607edcaa6&afSmartRedirect=y&gatewayAdapt=glo2usa4itemAdapt&_randl_shipto=US)

Measuring my heatset inserts [initeq @ Amazon](https://www.amazon.com/gp/product/B077CJV3Z9/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1), I get: `M3 L3.8 D5.5`, which does NOT match the sourcing guide's spec.

I'll add this to my list in the OP...

BerzerkaDurk, 2022-10-13:
Yes. The inserts in the Amazon link are very clearly NOT the correct insert. Somebody goofed the link.
Just checked the McMaster-Carr link, and those are incorrect as well.
[image]