VoronDesign/Voron-Stealthburner #45: M3x6 FHCS (anit squish thingymajig) needs threadlocker to prevent coming loose over time

pajtaz, 2022-10-04:
In the Stealthburner assembly manual there is no mention of using threadlocker to secure the M3x6 FHCS used for anti squish thingymajig. Multiple people have reported that screw coming loose over time, including myself. Using threadlocker on this screw should prevent the bolt from coming loose.

I suggest to add a note to use threadlocker when installing this screw. It can be done on page 20 and/or 27 of the assembly manual.

BerzerkaDurk, 2022-10-04:
I put Loctite 222 on mine. Bought it specifically for this application.

LuckyTurtleDev, 2022-10-04:
With FHCS did you mean? The screw on [page 20](https://raw.githubusercontent.com/VoronDesign/Voron-Stealthburner/66831c984ab8d9dfe8dfd114ca31e2aa4bee8c35/Manual/Assembly_Manual_SB.pdf#page=20) or the both on [page 56](https://raw.githubusercontent.com/VoronDesign/Voron-Stealthburner/66831c984ab8d9dfe8dfd114ca31e2aa4bee8c35/Manual/Assembly_Manual_SB.pdf#page=56)?

BerzerkaDurk, 2022-10-04:
> 

There's a clue in the title.

LuckyTurtleDev, 2022-10-04:
ups i am stupid

nemgrea, 2023-05-05:
https://github.com/VoronDesign/Voron-Stealthburner/pull/88

closed 2023-05-05 by nemgrea