VoronDesign/Voron-Stealthburner #50: Pg 23 of manual, words unclear

subatomicglue, 2022-10-07:
On Page 23 of the [voron stealthburner manual](https://github.com/VoronDesign/Voron-Stealthburner/raw/main/Manual/Assembly_Manual_SB.pdf)

We see:
```
DON’T OVERTIGHTEN
Tighten until the plastic bends and cracks.
Back up 2 turns, discard parts, reprint and try
again.
```

This language seems confusing to me.
1.) it tells me to not over tighten.  ok.  got it!
2.)  it then tells me to tighten until I bend and then crack the plastic.    (what?  this sounds bad)
3.)  then it tells me to back up 2 turns (turn the screws in reverse by...   how much?   2x  360deg rotations?)
4.) and to then reprint and do this again. (was I supposed to remember the position of the screw for next time?)

it sounds like i need sacrificial parts, and that cracking my parts is part of calibrating the screw tightness or something.

If I had to rewrite this, I would say instead:
```
DON’T OVERTIGHTEN
If you bend or crack the plastic, discard parts, reprint and try again.
```
Which is much more clear.

Or, in the scenario of calibrating your tightness using sacrificial parts, I'd say something more like this:
```
DON’T OVERTIGHTEN
Here's how to calibrate your tightness.   
Have 2 sets of parts ready, one is sacrificial:
Tighten until the plastic bends and cracks.
Back up your screw 2 turns, note the depth of your screw, discard parts, 
reprint and screw to the previously found depth.
```

As it is now, I can't tell what is being asked.  But I'd default to not cracking my parts if possible.  :-)

raymondh2, 2022-10-07:
It's a joke. Not actually directions

subatomicglue, 2022-10-07:
maybe needs a `(kidding!!)` after the writing?  I was baffled.  haha

BerzerkaDurk, 2022-10-07:
🤦‍♂️

closed 2023-01-24 by nemgrea