VoronDesign/Voron-Stealthburner #58: The stealthburner manual says to use M3 bolts when M2.5 bolts are correct

oernster, 2022-11-23:
Page 42 of the stealthburner manual says to use M3x8 bolts.  a) the holes are not big enough and b) they do not fit a dragon high flow hotend.  Also, the dragon high flow hotend needs to have the top piece removed which is not mentioned in the manual at an earlier point in the manual.

For me, I heat setted M3x8 bolts and THEN discovered that the hotend didn't take the bolts and completely damaged the parts trying to heat remove them.  Now I've had to resource the parts and I've learned the hard way that M2.5x8 is ACTUALLY required.

Please can you address this in the manual.

oernster, 2022-11-23:
OK, so let's clarify, perhaps the manual was accurate but perhaps the manual ought to indicate a dragon high flow as an option.  OR say careful the bolt sizes are only for the revo.

BerzerkaDurk, 2022-11-23:
>  OR say careful the bolt sizes are only for the revo.
[image]

Kosh42, 2022-11-25:
You heat what the bolts? 🤨

raymondh2, 2023-01-02:
Manual already addresses other hotends needing other size bolts

closed 2023-01-02 by raymondh2