VoronDesign/Voron-Stealthburner #62: Carriage Manual S.61 Wrong Length Bolts

thomasfjen, 2022-12-16:
[image]

Manuals says M3x30, i had to use M3x40. CAD also measures 40mm.

jonesuki, 2023-01-30:
On my MGN12 X-carriage, I can confirmed that M3x30 SHCS is long enough. I checked the STL files and the actual screw length required is about 27.x mm (because the hex bolt actually "sinked" inside the printed parts.
I discovered this because I accidentally installed M3x40 SHCS and both screws sticked out. The bottom one even triggered the X-endstop a few mm too soon.

raymondh2, 2023-01-30:
Screws are both 30mm. Manual is correct. 
[image]

thomasfjen, 2023-01-30:
[image]
You guys are correct, i was referring to the switchwire carriage and oversaw that this already covered in the switchwire manual.

raymondh2, 2023-01-30:
Ah it's the switchwire carriage. We should make note of this in the Stealthburner manual.

massis87, 2023-04-27:
The Trident manual actually mentions a 40mm bolt, which was so long in my case that it stuck out on the right side. I had to change it to a 30mm.

nemgrea, 2023-05-05:
https://github.com/VoronDesign/Voron-Stealthburner/pull/88

closed 2023-05-05 by nemgrea