VoronDesign/Voron-Stealthburner #68: trapped heat causing melt distortion issues for some users

Badnoob-327, 2023-01-04:
Some users have been experiencing melted carriages when using stealthburner

It may be limited to Dragon hotends (or it may not)
It happens even when using a sock.
It happens on a variety of ABS filaments.

I'm going to try and run a few experiments to confirm/quantify the issue, and then work on some fixes.

If you've had an issue where the M3x50's have pulled out their heatsets, this is the place for you. to post pictures of what you observed.

Badnoob-327, 2023-01-04:
from discord user "Carlos Diegues"
[image]
this appears to show significant and uniform deformation in Z

Badnoob-327, 2023-01-04:
from discord user"Pajtaz"
[image]

ScottSmall, 2023-01-04:
Tap bhcs shelf contacts, mediocre print to start with. Dragon ST, ~100h, socked, at 250-260
[image]
[image]

ScottSmall, 2023-01-04:
Tap bhcs shelt contacts, 200h Dragon HF, 260-270, socked, kvp brg likely printed at 250 (yes, with elephants foot). In both cases, the A side resting lug is a little more warpy.
[image]
[image]

pajtaz, 2023-01-04:
Stealthburner with Rapido and Tap. After roughly 100 hours of printing in 50 to 60 deg C chamber, printing one filament at 245 for 60-70 hours, and another filament at 260 for 20-25 hours, Tap Center and Tap Front softened. Photos are attached. Filament was FormFutura TitanX printed at 250 with Voron V0.1.
[image]
[image]

pajtaz, 2023-01-04:
After many hours printing in 50+ deg C chamber with Dragon HF, filaments with print temp ranging from 245 to 265, there were multiple softened X carriages and rear hotend mounts. Both were printed with 3 different filaments: TitanX (ABS+), ApolloX (ASA), EasyFil ABS (I guess just ABS, maybe slightly altered, should have softening temp at 103).

Softened X carriage, softened rear mount - 2 times since before 13/05/2022 (mentioned already on 23/04/2022)

After this happened, modified PRINT_START routine to move toolhead during soaking to max Y location, 100 mm above the bed, turn part cooling fan to 100%, and hotend to 50 deg C during heat soaking.

It did not help. Eventually both X carriage and rear hotend mount softened. Unfortunately I do not have a photo of softened X carriage but it was always softening at the bottom where the 2 M3x50 bolts are threaded. It was softening towards the center.

Eclypse3D, 2023-01-07:
Hatchbox abs printed @250c.  Chamber temp  is around 50-55c normally while heat soaked.  About 200hrs printing.  Had to retighten center rail 4 times already and as time went on started getting x axis waves in prints and now also lines on the Z, which I have already gone over everything.  I have replaced printed parts today will monitor most not all the waves are gone, lines have persisted.  Never had lines like these before..

[image]
[image]

Badnoob-327, 2023-01-08:
> Hatchbox abs printed @250c. Chamber temp is around 50-55c normally while heat soaked. About 200hrs printing. Had to retighten center rail 4 times already and as time went on started getting x axis [...]

what hotend are you using?

Badnoob-327, 2023-01-08:
> Stealthburner with Rapido and Tap. After roughly 100 hours of printing in 50 to 60 deg C chamber, printing one filament at 245 for 60-70 hours, and another filament at 260 for 20-25 hours, Tap [...]

the fact that this is also happening for Rapido for you tells me there may be something else contributing to it, as Rapido's mount does allow some airflow
[image]

pajtaz, 2023-01-08:
I will update tomorrow after another inspection of Tap and hotend mount but I think the Rapido mount is already warping. Give me until tomorrow and I will provide photos.

Badnoob-327, 2023-01-08:
the fact that you are getting warping all the way to the center part of tap indicates this likely isn't something we can fix with a design change to SB ducts.  Most likely you'll have to either reduce chamber temp or find a material that can withstand your higher temperatures.

pajtaz, 2023-01-08:
> 

I do not want to change chamber temps. It is ok if it is not possible to alleviate this problem, as long as that is agreed. I can then focus on printing some of those parts out of a different filament.
I will report back in a few days anyway to document if any Tap parts or Rapido rear hotend mount have warped since I last printed them.

Eclypse3D, 2023-01-08:
I am running the Revo Voron, which has been doing well till after I installed TAP.  Everything was been relatively fine, then about a week after I started seeing changes in prints, then it seemed like no matter what I did I started getting off prints that I never got before, my stacks used to be mint, no lines, no waves, no patterns....  Then I started noticing there was a noticeable amount of X axis play about .5mm and I disassembled everything and found that the center rail was loose.  I snug that down and a week later same thing, so I pulled it all off and found it was warping.   (To be fair I was printing alot of ABS over the past few weeks so chamber temps were at a constent.  (I also noted the rear hotend mount where it meshes with the front on the right side had flared outward too.

I have completely rebuilt TAP using the supplied printed parts from Fabreeko and the wavy part is way reduced, I can still just kinda see it but....  I replaced the 50mm rail and well so. 

Right now its running ok, but the print quality is not what it was, and I have already tore everything down to the belts, and re-racked etc.  Everything is tight, and all steps are what they should be.

I have a rapido coming tomorrow..  So I will be recalibrating and installing a complete new print head...(all printed parts etc.  Including extruder is rebuilt.)

Eclypse3D, 2023-01-08:
> 

I am running the Revo Voron, which has been doing well till after I installed TAP. Everything was been relatively fine, then about a week after I started seeing changes in prints, then it seemed like no matter what I did I started getting off prints that I never got before, my stacks used to be mint, no lines, no waves, no patterns.... Then I started noticing there was a noticeable amount of X axis play about .5mm and I disassembled everything and found that the center rail was loose. I snug that down and a week later same thing, so I pulled it all off and found it was warping. (To be fair I was printing alot of ABS over the past few weeks so chamber temps were at a constent. (I also noted the rear hotend mount where it meshes with the front on the right side had flared outward too.

I have completely rebuilt TAP using the supplied printed parts from Fabreeko and the wavy part is way reduced, I can still just kinda see it but.... I replaced the 50mm rail and well so.

Right now its running ok, but the print quality is not what it was, and I have already tore everything down to the belts, and re-racked etc. Everything is tight, and all steps are what they should be.

I have a rapido coming tomorrow.. So I will be recalibrating and installing a complete new print head...(all printed parts etc. Including extruder is rebuilt.)

Badnoob-327, 2023-01-08:
We tested this using 50-60C chamber temps at 23 hour/day print cycles, on 8 printers, for weeks though.  and none of those printers experience rail loosening or part warping.  Revo Voron is just about the best hotend we tested in terms of low waste heat.  

Its a mystery.

Eclypse3D, 2023-01-08:
> We tested this using 50-60C chamber temps at 23 hour/day print cycles, on 8 printers, for weeks though. and none of those printers experience rail loosening or part warping. Revo Voron is just about the best hotend we tested in terms of low waste heat.
> 
> Its a mystery.

The variable here is obviously the filament.

Badnoob-327, 2023-01-09:
theres other variables too.  thermistors can be off by as much as 10%, and aren't recording conditions away from the sensor.
operating procedures vary, for example preheating at Z=0 with fan off is different from preheating at Z=50 and 100% fan.
and others.

pajtaz, 2023-01-09:
[image]
After about 75 hours of printing in same conditions as before, rear hotend mount of Rapido warped. I attached a photo. This one was printed out of FormFutura Easy Fil ABS, which has a higher softening temperature than TitanX.

Tap parts printed out of same material did not warp this time.

paulsgale, 2023-02-03:
I have also experienced this twice now. The first time was a little while after I first built my 2.4r2, just as r2 was being released. I posted pics to discord but not sure if I have them here. 
Yesterday, I decided to perform some maintenance and removed the SB to check what I had noticed as a slight variation in my z height caused I think by a shift in the nozzle. I discovered that the heatset inserts had once again been pulled from their holes as the X carriage has softened and warped slightly. 

After the previous exact same issue, I was really careful not to over tighten the lower long screws, just pinching them tight but no more. It would seem that just from printing and the stresses that these have pulled free. I’ve not had any nozzle strikes and my printer is very finely tuned and looked after. 

Someone mentioned that some Formbot kits had been supplied with incorrect heatset inserts but I’ve just measured the two that have failed and they are perfectly in spec (M3x5x4)

It would help if the inserts could be mounted on the back side of the x carriage although I realise this would require even longer screws and might get in the way of other fastenings. 

My specs:
2.4r2 300mm
Parts printed from TitanX ABS (very good quality prints)
Heatset inserts seemed to fit well and were done with a heatset tool on my weller temperature controlled soldering iron (only hot enough to fit, no more)
Total printer hours since built - 261hrs
I print mainly eSun ABS+ at 240/250C
250C is the hottest I’ve ever printed
Chamber is at 42C max after several hours, no higher
Bed at 110C

[image]
[image]
[image]
[image]
[image]

paulsgale, 2023-02-03:
Oh, this is one of the heatset inserts that came loose. Measured with my digital calipers and seems in spec. Certainly exactly the same as another batch I bought from a different supplier. 
[image]

paulsgale, 2023-02-03:
One question though - the heatset insert hole is 10mm deep but I’ve fitted them flush to the surface. It might be that if I fit them deeper, this might help as they have 5mm extra clearance away from the heat source. Maybe that won’t make any difference though. Any thoughts on this?

paulsgale, 2023-02-05:
I stripped the print head today and fitted a newly printed x carriage. On inspection, I did find a fair bit of warping to the rear of the hot end mount too (sorry, forgot to take pictures but can get then if really needed). 
I guess the question is - why aren’t more users seeing this or is it that they aren’t stripping down very often and just don’t notice? Or is it that it’s purely something we’re doing?

my print head always parks about z=15mm rear left so can sit there for some time while heat soaking and before/after prints. Is the environment just too warm and maybe other users have different holding locations and positions? I’ve adjusted my macros to give a lot more distance from the bed and will see if that helps. 

One other question - would running the part fan at a low level help while heating/waiting etc? Does the part fan airflow even get up into the areas affected at all r is that purely the HE fan?

Badnoob-327, 2023-02-06:
you should have your part fan on during heatsoak. and we suggest more space than 15mm, 30-50 would be better.

pgale, 2023-02-06:
I'll modify my macros - I've already increased the height during heat soak. Does it matter what fan percentage or rather would you recommend a minimum?

pajtaz, 2023-02-06:
Keep it simple: run the part cooling fan at 100% when you are soaking and set the hotend to something like 60 C so the hotend fan runs also. Move the toolhead 100 mm above the bed.

This will not eliminate the issue. Eventually there will be softening due to the heat being trapped there while printing in a hot chamber. However it can help reduce the softening that happens during soaking.

pgale, 2023-02-06:
Thanks. Will do that.

Gi7mo, 2023-04-01:
[image]
dragon hotend, after less then 100 hours of print
heatsoak happens behind the bed 15mm above the purge bucket
my afterburner carriage was deformed too, but not that bad and after hundreds of print hours

ekhaliul, 2023-05-24:
I am seeing some melting as well. In my case the heat inserts would fall out when taking things apart. More pics here as well https://imgur.com/a/9kWMeLz. (following up from this thread [reddit](https://www.reddit.com/r/VORONDesign/comments/13pdzxg/is_there_a_fix_for_falling_out_threaded_inserts/))
I have enclosed chamber with 40-45C inside during print. Also print with heated bed @100C. I mostly print PLA/ABS/PETG with hotend temp 260C max, Mosquito hotend.
[image]

Badnoob-327, 2023-05-24:
> ith 40-45C inside during print. Also print with heated bed @100c. I mostly print PLA/ABS/PETG with hotend temp 260C max, Mosquito h

Mosquito pumps out a lot of heat. What kind of plastic is that printed out of? ABS? ABS+? what brand?

ekhaliul, 2023-05-24:
This was printed with ABS, hatchbox brand.

Good point about charring. I am running nevermore inside enclosure, but I will need a more comprehensive solution to prevent charring in the first place.

joaquinabian, 2023-06-03:
Got problems. Inserts are also loose.
Only a few printing hours (25 max) at 250-255/110, chamber between 45-55.
Dragon HF + CW2 at 250 mm/s 0.4 nozzle, voron settings
The parts are PIF, printed on TitanX for a Trident 300.

[image]

pfielding, 2023-07-03:
I had the heatsets pop on my carriage twice now.  I don't have photos of it pre-fixing unfortunately, though if it happens again I'll definitely take a couple of shots before doing anything.  The carriage itself seems mostly fine, it's just the two heatsets that hold the lower front plate through to the carriage that came loose.    Running a Dragon HF, the heatbreak fan seems to be pushing good air.   I mostly print ABS at about 250/110, with a chamber that sits around 58-60.

CrappyGamingXD, 2023-07-05:
Running dragon hf and yep, I'm getting this issue. Pretty bad too, my heat sets dont stay in for long. Would like to see a fix of some sort if possible.

TuningJohn, 2023-09-15:
We have experienced the same issue several times now. The threaded inserts become loose and pull out, causing the toolhead to become floppy.  We have used a couple different threaded inserts at this point, the last try used a longer version than the stock ones. We made sure to keep the inserts flush with the carriage mount front surface, but it still failed while running a Dragon HF around 250 C. Internal chamber temps are around 42 C.

## Root cause:
The threaded inserts are too close to the hot end and the plastic around them loosens. The vibration from toolhead movement causes the joint to become loose, the inserts eventually pull out, and printing become unreliable.

## Recommended fix:
In my opinion, no change in the threaded inserts will fix the problem, they must be relocated somewhere further away from the hotend. My first thought was to use longer screws and put the threaded inserts (or just a nut) on the _back_ side of the carriage mount part rather than the front. The hole for these two screws will go all the way through the part. This way the inserts are not pulling OUT of the hole, but rather pulled INTO the hole from the back side. 

I would make the fix myself, but I use Solidworks and can't seem to get the STEP file of the full assy to import. Can anyone send me just those two step files? I am concerned the horizontal screw that holds the two carriage parts together will interfere with this solution.

## Picture
[image]

pajtaz, 2023-09-15:
Let me add to this. I recently used Tap with Xol2 toolhead and on that toolhead I initially had to use M3 bolts from the front going into M3 inserts in the Tap carriage. However very quickly those M3 bolts were loose. Inserts themselves were ok but no matter how much I tightened (parts are also PCCF) the bolts came loose.

Recently I switched to the Xol toolhead. In the version I am using there are no inserts in the X carriage but there are M2.5 inserts in the hotend mount. M2.5 bolts go from the back of the X carriage into the hotend mount. I have now used that configuration for many hours in 77C to 80C chamber and even after decent speeds and accels, the bolts did not come loose and the heated inserts also did not come loose.

Therefore I would suggest to put the heated inserts into the back of the hotend mount and leave just a hole in X carriage/Tap carriage. Then use M3 bolts from the back of the X carriage/Tap carriage to secure the hotend. Maybe smaller bolts can still be used to secure Stealthburner main body to the hotend mount by putting another set of heated inserts in the front of the hotend mount.

wineds, 2023-11-13:
I have had a dreadful time with this issue.  Printer is Voron 2.4 REV C LDO kit with Voron Revo.  Original toolhead parts were PCCF but I have also tried ABS and PA11 Nylon CF.  I mostly print ABS at 250/110 C.  Since the V2.4 completion a few months ago I have been battling drifting Z offset.  Always in one direction.  Dissembled the toolhead and noticed heatset inserts were pulling out of the bottom (and to a lesser extent the CW2 mounts at the top).  Tried and ABS X carriage, same issue.  Tried a PA11 Nylon CF X carriage, same issue.  Tried Swapping the Omron inductive probe for Euclid but still drifting Z offset.  Eventually tried both printed and CNC TAP which resolved the issue.  Note that unlike some here I don't have X carriage distortion.  
TAP is awesome but my preference was to run a Euclid probe.  I had some Square M3 nuts on hand so I decided to try to modify the X carriage STLs to suit.  I have been running a modified PA11 Nylon CF X carriage and the square nuts seem to have resolved the issue.  I did also print ABS versions but have not yet tested.  
The square nuts are a tight fit in ABS, less so in PA11 Nylon CF.  When pressing them in be sure to start them straight then use a flat surface to force them flush with the X carriage surface.  You can then use a small flat head screwdriver to insert them to the correct depth.  If you overshoot you can use a small alley key to correct. 
Square nuts :  https://au.element14.com/tr-fastenings/m3-sqst-z100bs4183/press-nut-square-bzp-m3-pk100/dp/1419469?ost=141-9469  They also seem to be available from Newark/Farnell globally.  

[image]
[image]
[image]

vajonam, 2023-11-16:
I had some similar issues. Voxelabsb ABS Pro. Usually print ABS only 265 to 270. I have just reprinted the parts. Considering going for aluminum X Carriage  

[image]

[image]

[image]

sarkar-anwar, 2024-01-25:
Similar issue. This is the 2nd time. This time it was printed using 3dxtech ASA.

[image]

Zoro189, 2024-02-01:
I have same isue, PIF parts so there shouldn't be any issue with print quality. Lost heat insert and also hotend can be slightly rotated. I have dragon HF mounted on it. And I print mostly to this point only PLA on printer, so no high chamber temps. I maybe went trough 400g of ABS so far
[image]

Zoro189, 2024-02-26:
Some new pictures, same stealthburner from comment above 
[image]
[image]

archi, 2024-07-24:
And another case here. This time with the Dragon UHF mount.
A design error in that mount accelerated this for me and caused the heatsets to survive only about 3h of ABS printing.
Additionally, I tightened the bolts really well (for "some reason" they always came loose, causing degraded IS results).

For the design error, I pinged the designer of the mount on Discord and we talked about it in in my build log.
See the message around here: https://discord.com/channels/460117602945990666/1260216015208448120/1265594856193921067  

I first used Voron Tap, which survived it better (75h, give or take, also ABS printing). They have the inserts on the rear, but those were also pulled quite far into the plastic.

Some epoxy allowed me to fix my failing carriage and reprint the X carriage (the one with the UHF-compatible Cartographer mount from printables - https://www.printables.com/model/925457-beacon3dcartographer-probe-stealthburner-volcanouh). The inserts sit now much tighter, probably because didn't calibrate dimensional accuracy of ABS, yet (so all holes were to tight; a blessing in this case).

My ABS temps are 260C/105C/40-55C; I used the silicone sock, but had it the wrong way around; I doubt this had a large impact.

That concludes the anecdata from me. Images in the linked thread (but not from the failed X carriage - the holes were the inserts sat were gigantic though).

My very limited experience with these things indicates there is just too much heat with these hotend. Maybe moving the inserts to the back of the carriage and using 25mm longer screws solves this, maybe not. I'll try to reprint the parts in a more heat resistant material and eventually move to a CNC carriage. IMHO these are the most sane solutions, the ABS carriage will surely fail again (I'm sure to add the hints to my PRINT_START and PRINT_END, but they're just a stop gap). But there is always a next revision, and except for this setback I'm pleased with SB! So thanks for the huge effort you put into this, and keep up the good work :)

Saberjag, 2024-08-09:
I've been fighting this issue since March of 2023 and thought I was alone with this.  I've been replacing my x carriage every 3 months and sometimes gluing in my threaded inserts to get by to print replacement parts.  My klickly probe mount also needed to be replaced every 3 months.  

To fix this, I just upgraded to CNC x carriage.  
https://github.com/Vitalii3D-xyz/VoronLightweightXCarriage/

I had to redesign a mount for PCB klickly.  

[image]
[image]

pgale, 2025-10-19:
> Hi, I am the another one. I just reported it on voron [forum](https://forum.vorondesign.com/threads/stealthburner-melted-parts-around-inserts-inserts-pulled.12068/). Siboor kit 2.4. Is the CNC carriage the only way to solve it? 

I moved to a cnc carriage a while ago. Even with CF/ASA printed parts before that, I still saw a softening around the inserts and them eventually pulling out. I guess it depends how tightly you screw into those as to whether you see them pulling out completely. I definitely think there's an issue.

pgale, 2025-10-19:
Not sure about clicky, you'd have to check the info from the cnc part you want to use. I use the Chaoticlab CNC Voron Tap part and it works very well.

jlaustill, 2026-02-09:
> I have had a dreadful time with this issue. Printer is Voron 2.4 REV C LDO kit with Voron Revo. Original toolhead parts were PCCF but I have also tried ABS and PA11 Nylon CF. I mostly print ABS at 250/110 C (chamber ~50C). Since the V2.4 completion a few months ago I have been battling drifting Z offset. Always in one direction. Dissembled the toolhead and noticed heatset inserts were pulling out of the bottom (and to a lesser extent the CW2 mounts at the top). Tried an ABS X carriage, same issue. Tried a PA11 Nylon CF X carriage, same issue. Tried Swapping the Omron inductive probe for Euclid but still drifting Z offset. Eventually tried both printed and CNC TAP which resolved the issue. Note that unlike some here I don't have X carriage distortion. TAP is awesome but my preference was to run a Euclid probe. I had some Square M3 nuts on hand so I decided to try to modify the X carriage STLs to suit. I have been running a modified PA11 Nylon CF X carriage for about 3 weeks now with a few hundred hours printing and the square nuts seem to have resolved the issue. I did also print ABS versions but have not yet tested. The square nuts are a tight fit in ABS, less so in PA11 Nylon CF. When pressing them in be sure to start them straight then use a flat surface to force them flush with the X carriage surface. You can then use a small flat head screwdriver to insert them to the correct depth. If you overshoot you can use a small alley key to correct.
> 
> Square nuts : https://au.element14.com/tr-fastenings/m3-sqst-z100bs4183/press-nut-square-bzp-m3-pk100/dp/1419469?ost=141-9469 They also seem to be available from Newark/Farnell globally.
> 
> [image] [image] [image]
> 
> [STLs.zip](https://github.com/VoronDesign/Voron-Stealthburner/files/13331219/STLs.zip)
> 
> Note : You will need to use M3x12mm SHCS to mount the CW2.

OMG Thank you!!!! This is the way it should be, accidentally tighten to much once, replace the nut, done

thunderkeys, 2026-11-07:
Alternate X carriage https://github.com/VoronDesign/voron-chariot/ addresses the majority of these issues with all heatsets being pull-through, as well as some beefed up rear mounts on the ducts to guard against melting. Closing issue.

closed 2026-11-07 by thunderkeys