VoronDesign/Voron-Stealthburner #70: Clockwork 2 Main Body bearing bore appears too large

cmgreyhounds, 2023-01-09:
When assembling my Stealthburner today I found that the bearing for the Bondtech gear was loose in the Main Body. I pulled the stl into Fusion 360 and checked and it looks to be around 8.20 mm, but the bearing is 8.00 mm. That is alot of play right next to the filament feed groove and could cause some amount of inaccuracy in extrusion.

nemgrea, 2023-01-18:
the cad dimensions and the actual print dimensions are not the same. this is intentional to account for ABS shrinkage and the FDM process

closed 2023-01-18 by nemgrea

cmgreyhounds, 2023-01-18:
That makes sense, but why is the bore in the Main Body bigger than the bore in the Motor Plate? 

If they both carry the same size bearing and are both made of ABS shouldn't they be the same size?