VoronDesign/Voron-Stealthburner #71: [documentation] Mention access to 50t

boop5, 2023-01-14:
Page 30 is about setting backlash.

It could be helpful to mention that you can actually access the 50t to test the backlash. That would never have occured to me when I built my first voron (first printer).

[image]

On a side note: It'd be cool to get a link to some youtube video or w/e similar to the heat set insert qr code

[image]

neeamradia, 2023-01-15:
I agree completely, for people like me who have never worked with this kind of stuff before, this page was very confusing as it says 'slight play' but as I have never experienced this before, I didn't know how much play is required and 'slight' doesn't really help. Even 4-5 months after building the printer i have no idea!

nemgrea, 2023-05-05:
https://github.com/VoronDesign/Voron-Stealthburner/pull/88

closed 2023-05-05 by nemgrea