VoronDesign/Voron-Stealthburner #77: TOOL CARTRIDGE Screws (Page 41)

SharkyMoto, 2023-02-11:
The manual calls for M3x12mm screws here, but when using the Rapido cartridge, those are too short (0-1 Threads engagement), 16mm works great though, so maybe worth mentioning it in the Manual.
[image]

stahlfabrik, 2023-02-16:
Same is true for the revo voron! I came here to see if there is an issue for this. The 12mm would not engage with the thread for me.

stahlfabrik, 2023-02-16:
Ah no - one headset I pushed to far in... the other is ok with 12mm

massis87, 2023-04-27:
I agree, I had the exact same issue. My inserts were pushed to just proud of the service and then leveled by cooling them down pressing onto a polished granite surface, so they're completely flush, yet they would not engage. 16mm worked perfectly.

nemgrea, 2023-05-05:
https://github.com/VoronDesign/Voron-Stealthburner/pull/88

closed 2023-05-05 by nemgrea