VoronDesign/Voron-Stealthburner #82: Part cooling fan - unbalanced airflow

btthrt, 2023-03-16:
Hi guys

Decided to move from AB-BN to Stealthburner.

The first thing I've noticed is that part cooling airflow is now not even close to uniform. 
With AB-BN it was around 60-40, with stealthburner it's more 90-10 which results in uneven cooling and yada yada yada.

Tried with 2 fans: original sunnon and gdstime. No luck.

btthrt, 2023-03-16:
Fan setting: spin-spin to push the air in
Toolhead: any
Photos: posts pictures of air

Also, both gdstime and sunnon rotors are not centered on the inlet. This known issue can potentially contribute.

How to reproduce the main issue: 
- keep toolhead attached or not, does not matter
- put a glass of water close to the print head nozzle, right between the fan ducts
- turn part cooling fan on
- observe water being pushed from left to right, since there is no air pressure in the right fan duct

raymondh2, 2023-03-17:
What toolhead are you using?

btthrt, 2023-03-17:
Does it really matter? They all have the same internal structure.
I've tried dragon, revo and rapido toolheads, all perform the same. 
My guess - this whole problem is due to stealthburner's intake design, especially how the  part cooling fan is seated. (!!!) Not the toolhead design. 

If you place your hand near the inlet, you can feel that the fan is actually pushing a substantional ammount of air back. 
My guess is, again, this is due to the rotor being off-centered and due to enormous gap between the rotor and the intake.
[image]

closed 2023-04-03 by raymondh2