VoronDesign/Voron-Stealthburner #85: Part "pcb_spacer" and the two related screws not in manual

miracuruzrh, 2023-04-23:
Hi

I have not seen any mention of the red "pcb_spacer" and the two screws in the manual.
Maybe they should be added in the manual. I'm not 100% sure if it's correct. It seems to me that M3x6 screws should be used.
Then it would be 1.8mm clearance to the bottom of the hole, where the insert is screwed in. Is this correct?
A M3x8 screw would not be able to be tightened completely, as the screw would touch the bottom of the hole and the "pcb_spacer" would have 0.2mm clearance. Or in other words, the screw is 0.2mm to long if a M3x8 would be used. Hopefully that's correct.
I have attached a drawing as well.
[image]
[Clockwork2.pdf](https://github.com/VoronDesign/Voron-Stealthburner/files/11302897/Clockwork2.pdf)

BerzerkaDurk, 2023-05-05:
Side note: it doesn't look like you accounted for the 1.6mm board thickness in your model.

[image]

nemgrea, 2023-05-05:
ah ok, ill re-open this but list it as a CAD issue then instead i think that fits better than a manual tag

BerzerkaDurk, 2023-05-05:
> ah ok, ill re-open this but list it as a CAD issue then instead i think that fits better than a manual tag

Sorry, nemgrea. I didn't mean you. I was directing that at OP who said that the M8 was too long, but it doesn't look like they included the board thickness in their calculation.

closed 2026-11-07 by taco-niemand