VoronDesign/Voron-Stealthburner #99: X Carriage Hardware

1John419, 2023-07-11:
In the Voron Stealthburner Manual (2022-06-30), on page 61, a M3x40 SHCS is indicated on top and a M3x30 SHCS on bottom of the X Carriage halves. This is also reflected in the Voron step file.

I may be completely wrong (this is my first build), but I printed these parts and find that a M3x30 SHCS is appropriate for both top and bottom. This is also reflected in the STL files (x_frame_V2TR_MGN12_left.stl and x_frame_V2TR_MGN12_right.stl). The depth of the hex nut recess is 9mm top and bottom. The socket head recess is 6.6mm top and bottom. This would also suggest the same screw top and bottom.

The same discrepancy appears on page 140 of the Voron 2.4R2 Build Guide (2022-07-04).

nemgrea, 2023-07-11:
fixed

closed 2023-07-11 by nemgrea

1John419, 2023-07-11:
So, the X Carriage STL folder has been removed completely. Does this mean that the V2.4 STLs should be used instead and that I need to reprint these parts (the right side is noticeably different)?

nemgrea, 2023-07-11:
the x carriages are on their way to the respective printer repos, you can pull them from the pending PR's or you can roll the SB repo back and get them from there. we're in the middle of replacing afterburner with stelathburner across all the repos