VoronDesign/Voron-Stealthburner #107: Part cooling is not symmetrical (X-axis)

maxim-kukushkin, 2023-07-22:
I've recently assembled my Voron Trident with Stealthburner (LDO kit) and I'm seeing issues printing PLA due to extensive amount of warping. The warping is not symmetrical and suggests issues with part cooling in Stealhburner's duct.

The hotend that I'm using is **TriangleLab Dragon HF**. However, given the placement of the cooling it doesn't matter.

The cooling duct vents are located in front (along Y-axis). And they blow at the nozzle at an angle. I.e., the air flow is symmetrical along Y-axis, however, it's not symmetrical along X-axis.

It can be clearly seen on the printed parts, where majority of warping happens at the back (opposite side from cooling vents). See the pictures below.

There's no air leakage. Also, the same PLA has previously been used on a super cheap Prusa i3 clone, inside and enclosure, but with symmetric cooling ducts pointing strait to the nozzle from left and right along X-axis and it was printing perfectly, both high speed and low speed.

So, what I'm trying to say here is that the problem is the shape and location of the air vents for part cooling, not the filament, enclosure, speed, etc.

Would you consider slightly changing the design of the hotend cartridge to direct the flow symmetrically along both X and Y axes? I think similarly to how it's done in Afterburner (meanwhile I'll try to downgrade my printer to Afterburner to see how that one performs). But to me it's obvious - if the air flow is not symmetrical, the effect of it will not be symmetrical too and I've already had a chance to confirm that in the past when trying different cooling ducts.

[image]
[image]

maxim-kukushkin, 2023-07-30:
For comparison, this is how the exact same file was printed using Afterburner (same printer, same hotend, same gcode, same closed doors, just different toolhead).

It still has some issues (especially in front), but it's not even comparable with Stealthburner's result.

This is because Afterburner uses fully symmetric part cooling and the air vents are directed right at the nozzle tip.

The reason Afterburner's result is not ideal is:
- too long air vents (they should be around 3 times shorter)
- less performant fan (4020)
Otherwise the 3DBenchy would have perfect overhangs.

This can be achieved with Stealhburner if it's slightly redesigned to direct the air flow differently (like explained above). It doesn't have fan power issues, it only needs cooling ducts to follow AB's design, but have slightly shorter vents.

[image]
[image]

Biffes, 2023-08-02:
Hi there,
i have some trouble with pla on the left side of benchy too. I did some testing an the airflow seems to be slightly off to the front of sb. Here are some pictures:

[image]
[image]

mdvorak, 2023-08-04:
I did not do any extensive tests, but I have same experience, cooling is not symmetrical. Overhangs are always worse from one side.
Using Dragonfly BMS.

nielssteigenga, 2023-08-05:
I'm experiencing the exact same issue. Made some video's with smoke added to visualise the airflow. I will post them asap.

maxim-kukushkin, 2023-08-26:
Just as a proof that it's possible to print PLA without warping in with the closed printer doors (without removing the acrylic panes, like many people suggest for printing PLA), below is my custom tool head based on Bondtech LGX Lite extruder.

The 3D benchy was printed on the same printer: same G-Code file, same hotend, same printer, same spool of filament, closed doors. My toolhead uses strong air flow directed **symmetrically** straight on the nozzle tip. See that there's no warping anymore.

[image]
[image]
[image]
[image]
[image]
[image]

maxim-kukushkin, 2023-08-26:
The above toolhead is work-in-progress and at the moment is in sort of POC stage. I'm planning to add a front cover and LEDs to it. After that I'm thinking to share it either on GitHub or Thingiverse, but no specific timelines yet

Biffes, 2023-10-22:
Hi, I just wanted to give you a small update.
I adjusted the ventilation ducts so that they extend behind the nozzle.
Unfortunately, this reduces the pressure significantly, so that too little air still reaches the part:

[image]

My current solution is that I have installed additional fans under the Y-axes. This solution works surprisingly well:

[image]

I used the "breeze" solution by jphalip ( https://www.printables.com/de/model/532824-breeze-auxiliary-fan-system-for-voron-24/files
 ) and modified it to 4020 fans with air ducts and used the multi_pin command to pair it with the parts cooler.

I still believe that the ventilation ducts of the SB can be optimized. Maybe it would help if you rotate the hot end fan 45 degrees and route the parts fan with a single air duct.

maxim-kukushkin, 2023-12-05:
It took me a while, but I finally managed to complete the design of the toolhead I drafted above and to write BOM and assembly instructions for it
It's now available both in form of OpenSCAD source code (GPLv3) and printable STLs. I've been using it for a couple of months now. Please, feel free to try it and I'd totally appreciate any feedback :)

Hopefully this idea can also be considered as proof-of-concept to support this Issue and hopefully improve the design of future Stealthburner's revisions (cause I'd really prefer to use a ready-made solution instead of spending a couple of months designing my own)

[image]
[image]