VoronDesign/Voron-Stealthburner #116: Rapido 2.0 thermistor wires block SB

gizzygizmo, 2023-09-29:
STLs in STLs/Stealthburner/Printheads/phaetus_rapido have been a bit of a squeeze on Rapido1.0 but were possible by clocking the output of the wires and bending them slightly.  On 2.0 it appears the thicker thermistor wires block the screw hole completely and you must egress the "front" as the "back" will block TAP.   If you egress the front you have to do some less than ideal u-turns and bending/shoving to get things in place.

gizzygizmo, 2023-09-29:
Attaching pics for reference...

[image]
[image]
[image]
[image]

arkeet, 2023-10-05:
I hear a revision of the Rapido mount is in the works, but in the meantime this is how I ran the wires on mine:

[image]

gizzygizmo, 2023-10-05:
I'll give it a try... i didn't want to push to hard on the thermistor wires but i supposed it's not the end of the word if it breaks off, since it's easy to replace now at about $15.   I still have my original rapido if it it should break or cause issues before a new STL is released.

Can probably shave down the hole a little bit too... the through-screw is probably what's giving the structure/support here.

louis-evard, 2023-10-17:
I've made a minor modification to the parts to at least allow the thermistor to fit without forcing the base... Had to cut some landing surface from the screw tower and expose the main the screw, but built some landing surface back in other areas. I have not actually tested this yet
[image]
[image]

gizzygizmo, 2023-10-17:
Can you share the STL?  I'm rebuilding my hotend now and can test it with you... redoing w/ canbus+rapido2 and can test this out.  Looks like a viable interim solution to me.

louis-evard, 2023-10-18:
[stealthburner_printhead_rapido_2.0_fitment_mod_rev1.zip](https://github.com/VoronDesign/Voron-Stealthburner/files/12983464/stealthburner_printhead_rapido_2.0_fitment_mod_rev1.zip)

Here's the stl's, I included a modified f3d file also - my AutoCAD skills are not the best... still learning.

gizzygizmo, 2023-10-24:
Thanks for posting, i was able to print this out an test fitment and it looks good.  It's not ideal, but i think this is much better than trying to cram the wires out the front and wrap them around.  Moving material to the side will def help with support and thanks for doing both front & rear.   It had not occurred to me that the other side would likely need some adjustment too.

I'm still assembling CW2 + Canbus so will report back in a month or two once i've tested it in production.  For now, here's a shot of mine printed and centered:
[image]

kyleisah, 2023-10-24:
Fixed with PR #118

closed 2023-10-24 by kyleisah

gizzygizmo, 2023-10-24:
> I'm not sure if I'm missing something here... I've printed the new PR 118 files, aligned the Rapido 2.0 grub screw thermistor and heater outputs but... I'm still unable to successfully mount the Rapido 2.0 in any orientation I appear to have to badly deform the thermistor output to an uncomfortable degree. Does anyone have any photos showing this correctly mounted?

Not sure either... i pulled the latest merge and see a new "2.0" dir with an STL that has a different MD5 than the original one... so they're technically different.  But importing them and overlaying on top of each other doesn't appear to show any diff in the mounting hole that blocks the temp sensor.

kyleisah, 2023-10-24:
@louis-evard https://media.discordapp.net/attachments/611278504914583562/1166478999011213322/IMG_0784.jpg?ex=654aa339&is=65382e39&hm=30f86d131e31a219bd2ddd9fdf872cd4b86df1c1704d7e883b875874171df121&

this is the intended installation method. The heater wires and strain relief need to be bent down slightly to clear the screw boss, and the thermistor is routed above the screw boss. 

The mounting holes were rotated a touch so the thermistor wires aren’t smushed up against the screw boss and have an adequate amount of room.

jagerman, 2023-11-18:
Is the intention here that the three screws attaching the heater to the heatsink have to be removed and the heatsink rotated to make this work?  I have two different brand new Rapdio 2's and neither match the photo without doing that adjustment.

gizzygizmo, 2023-11-18:
Yup... to use the PR/Merged rework you need to disconnect the top and rotate it so the set screw lines up matching the readme diagram here: https://github.com/VoronDesign/Voron-Stealthburner/blob/main/STLs/Stealthburner/Printheads/phaetus_rapido_v2/README.md

Full disclosure, i didn't like this solution and used the attached STL in this thread that chops the top of the screw hole and adds some material to the left.   Another user also proposed just sticking the wires out the front and doing a tight u-turn toward the back.   I'm sure all of the solutions work, i just chose the one that looked the most elegant to me.

mavera3, 2024-01-05:
> Yup... to use the PR/Merged rework you need to disconnect the top and rotate it so the set screw lines up matching the readme diagram here: [...]

The documentation is not correct though. I just went through this with my rapido v2. The picture you show does work however the clocking of the screw holes compared to the grub screw and thermistor wires in your picture (that works) does not match what is shown in the documentation. The documentation says "Take care to ensure you are using the correct orientation for the V2, the following image just explains the process." The problem that I think needs correcting is to define what (via text or a second image) what that correct orientation is. Readers are left to determine this on their own and I and others in this thread were interpreting the documentation to say that the orientation shown is how the v2 should be clocked which is not correct.

mshrem, 2024-01-15:
So i am running into issues trying to mount my hotend, so i do not have any printer and ordered pif parts , the part is for the rapido , however i have the rapido v2 is there a difference ? 

I am trying to do the steps provided in the guide for the rapido orientation however the only way that it works gives me a massive kiink on the spring wires. 

[image]
[image]

Is this is how am i supposed to be doing this ?

mavera3, 2024-01-15:
> So i am running into issues trying to mount my hotend, so i do not have any printer and ordered pif parts , the part is for the rapido , however i have the rapido v2 is there a difference ?
> 
> I am trying to do the steps provided in the guide for the rapido orientation however the only way that it works gives me a massive kiink on the spring wires.
> 
> ![rn_image_picker_lib_temp_63bba78f-94f7-4833-b469-c49107c9842d](https://private-user-images.githubusercontent.com/560713/296706744-4d6a8f6b-553a-4bbd-b6b0-04dcd7217a9b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUzMjQ3MzIsIm5iZiI6MTcwNTMyNDQzMiwicGF0aCI6Ii81NjA3MTMvMjk2NzA2NzQ0LTRkNmE4ZjZiLTU1M2EtNGJiZC1iNmIwLTA0ZGNkNzIxN2E5Yi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExNVQxMzEzNTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01NTBkMjZlZDM2NTg5ZmI5MzFmZTdiZTI0YTRmNmQwZDdhZTg3NDU2ZWIzMmM1MTZiNjNiMTJjYzM2YjJjODY0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.lzOO07Xa5LhbE5QliLOgEW5dW03V0Hq1gOiyLpaIk_Y) ![PXL_20240115_070527502 MP](https://private-user-images.githubusercontent.com/560713/296706758-954ba3d3-cb85-48f7-b6b0-1ba0c1477426.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUzMjQ3MzIsIm5iZiI6MTcwNTMyNDQzMiwicGF0aCI6Ii81NjA3MTMvMjk2NzA2NzU4LTk1NGJhM2QzLWNiODUtNDhmNy1iNmIwLTFiYTBjMTQ3NzQyNi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExNVQxMzEzNTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMzEwZjUyYTQxOTMwNjU3OTA1YjNiNDRkNGUwM2Q0NTZiZDZkZjc0MDM5ZTVkNWI1MTAyODAyMzhjZGY1YTY4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.SyYd1AB7kGe4os0eUkl4GnMPrcqnv-jIwwdjhUlPaEI)
> 
> Is this is how am i supposed to be doing this ?

I can't tell from your picture. Did you take off the hot side and rotate it so the grub screw is next to the thermistor and heater wires? Also the proper fitment doesn't have the wires flat against the printed part, they stick out some.  Like this picture. 
[image]

mshrem, 2024-01-16:
> I can't tell from your picture. Did you take off the hot side and rotate it so the grub screw is next to the thermistor and heater wires? Also the proper fitment doesn't have the wires flat against [...]

Thanks , well it seems that my pif parts only included are for the rapido 1 not the 2 , i also have the HF version dont know if that affects the mounting or not. 

So i asked my friend if he could print one for me since i do not have a printer capable of printing ABS. 

Here are the results , dont know if this is what should look like. 

[image]
[image]