VoronDesign/Voron-Stealthburner #124: Rapido 2 Toolhead: Orientation in README.md not correct.

Laikulo, 2023-12-23:
The supported orientation in the markdown file at STLs/Stealthburner/Printheads/phaetus_rapido_v2/README.md indicates that the grub screw should be at 4:30, and the wires at 12:00, with the screwhole nearest the wires at 12:30, however this orientation does not align the screwholes.

I found that the wires at 12:00, screw at 7:30 with the hole nearest the wires at 11:30 when viewed from the top (the face with the four holes as indicated.

Laikulo, 2023-12-23:
This is actually going to be user error. I misinterpreted the note about the orientation being different for the v2, and was unsure what figure to use. I just wound up testing all three orientations

closed 2023-12-23 by Laikulo

michmerr, 2024-03-24:
I just went through this.

I think the "supported" alignment is correct, with the wires at 12 o'clock and the grub screw at 4 o'clock. _However_, the [alignment image with the element highlighted in blue](https://github.com/VoronDesign/Voron-Stealthburner/blob/main/STLs/Stealthburner/Printheads/phaetus_rapido_v2/Rapido_V2_Assembly_Images/v2_heatsink_orientation.png) shows the grub screw aligned with the wires, which is not correct.