VoronDesign/Voron-Stealthburner #131: Missing main body CAD

Garethgb, 2024-02-24:
CAD files for all printheads and CW2 but cant find the main body CAD, only STL.

Bunny350, 2024-02-24:
Hi there,
It should be available in Stealthburner repository. If you need them, you can [click here](https://github.com/VoronDesign/Voron-Stealthburner/tree/main/CAD).

Garethgb, 2024-02-24:
> Hi there, It should be available in Stealthburner repository. If you need them, you can [click here](https://github.com/VoronDesign/Voron-Stealthburner/tree/main/CAD). _Proud owner and modder of two Voron printers._

Thanks. I was looking for it to be listed separately like the other parts but found it as the complete unit step file which is fine for what I need.

closed 2024-02-27 by nemgrea