VoronDesign/Voron-Stealthburner #132: Clockwork2: CAD / STL has insufficient tolerances - not printable.

dmradford, 2024-04-06:
The pieces [a]_guidler_a and [a]_guidler_b are supposed to interlock. However upon printing, they cannot be pressed together without damaging them. Inspecting the step file for Clockwork2, it becomes apparent why this is - they are designed with almost zero tolerance between the parts - 0.025mm to 0.05mm. This appears to be an oversight or misplaced decimal as other interlocking parts of the Clockwork2 use a 0.2mm tolerance, such as the interlocking pieces of the motor plate and the main body. I observed this same issue in other pieces of the Clockwork2 step file that are supposed to slide, not interlock, such as the interface between the shuttle and the main body, and the shuttle and the latch. These sliding parts should likely have a gap of 0.3-0.4mm to allow them to slide together smoothly (I got a smooth but snug fit with a tolerance of 0.25mm). Please advise if these should all be addressed here, or if I should create individual issues for each piece.

guidlers fit
[image]

motor plate and main body
[image]

stvptrsn, 2024-04-09:
The fit between the guidler halves is critical to keep the BMG idler parallel to the drive gear.  The current clearance is intentional and was the result of a long testing period where several revisions were evaluated among a large group of beta testers.  It is proven to be printable as evidenced by the number of Stealthburner/CW2 assemblies out there.  I've printed it countless times and although it has been tight when my print tolerances were off, even to the point of requiring me to check my settings (I was over extruding), it still goes together across a reasonably wide range.  All this said it is being discussed by the dev team to see if it makes sense to tweak it a little more.

dmradford, 2024-04-13:
Thank you for the insight. I now understand how tightly the parts are designed - even a slight interference fit. For injection molded parts in ABS, a +0.15mm interference is within range for an interference press fit. So printers capable of hitting 0.15mm tolerances, plus the 0.05mm gap, appears to add up. However that's per side. So to match a +0.15 interference fit using a printer tuned to within 0.15mm, the gap would need to be at least 0.15mm: 0.15mm * 2 sides, minus interference of 0.15mm gives you (0.15*2)-0.15=0.15mm gap between parts in CAD. To be fair, only the most highly tuned printers are capable of hitting tolerances of even 0.15mm, and yet with a 0.05mm gap between the parts, the printer would need to be capable of hitting ±0.05mm per part unless XY comp is used. This is 3x more precise than a very finely tuned FDM printer is likely capable of. Even CNC machines can struggle with these tolerances in metal. That's less than 0.002".

Knowing its meant to be an interference fit, I would still recommend increasing the gap to at least 0.1mm, and likely 0.15mm. I printed mine with a 0.2mm gap and it slid in without interference but was still snug enough to not have any play.

stvptrsn, 2024-04-13:
Thank you for your suggestion, we will update the parts if we determine it is required.

closed 2024-04-13 by stvptrsn