VoronDesign/Voron-Stealthburner #140: Wrong files

Aleksbbgk, 2024-07-17:
When you download files from this link: https://github.com/VoronDesign/Voron-Stealthburner/releases/tag/1.0 , motor plate and main body of cw2 are not compatible.
[image]
[image]

i hope pictures will help.

nursereese, 2024-07-25:
Yep, I came here after printing the parts and finding the main body pieces incompatible. 
The current release #98 includes the incorrect part. 
I only found the "#101 sb shroud got lost" through a Google search for the correct file.

aceofjokers, 2024-08-12:
Considering its taken so long for someone to fix this....here are the files if anyone comes looking.
[CW2 body and plate.zip](https://github.com/user-attachments/files/16580032/CW2.body.and.plate.zip)

thunderkeys, 2026-10-26:
since fixed/no longer an issue.

closed 2026-10-26 by thunderkeys