VoronDesign/Voron-Stealthburner #141: Release STL for stealthburner main body is the wrong file

kurtjcu, 2024-08-04:
file in Voron-Stealthburner-1.0\STLs\Stealthburner\[a]_stealthburner_main_body.stl is one of the printheads bodies. I had to export stl from fusion to complete the set.

aceofjokers, 2024-08-12:
Incase anyone else comes looking for the files.....
[CW2 body and plate.zip](https://github.com/user-attachments/files/16580039/CW2.body.and.plate.zip)

foleac, 2024-09-27:
I don't think motoro_plate.stl was wrong. It is identical to the one in the repo, but [Clockwork2/main_body.stl](https://github.com/VoronDesign/Voron-Stealthburner/tree/main/STLs/Clockwork2/main_body.stl) is wrong. It does not fit the motor plate. The one in the repo works.

thunderkeys, 2026-10-26:
since fixed/no longer an issue.

closed 2026-10-26 by thunderkeys